LucidVR/opengloves-driver #33: Arduiono uno and Driver lucidgloves has no suitable devices.

weegeeday, 2021-03-22:
I had no Arduino micro so i just used a Uno instead, the Potentiometer readings works fine on the Arduino serial monitor, but the SteamVR web console says there is no suitable device, i did notice that the config file had a different com port so I had changed to the one my Arduino uses, but still steamvr said that no device was suitable. I have no idea if this is because of this error: "OpenService failed (1060)", or the fact that I'm using an Arduino Uno or that I'm just having one Potentiometer connected for testing. SteamVR does say that an unexpected SteamVR Error happened and that its a critical error.

danwillm, 2021-03-23:
Hello,

Sorry about the issue you're running into.

We know of an issue that causes a crash on startup when no headset is currently plugged into SteamVR, it's going to be fixed in the next update but hasn't released yet - have you made sure that your headset is plugged in?

If it's still crashing, double check all of the potentiometers readings from the pins that aren't connected all read normal values and that your settings file is tied to the correct port for the arduino.

Hopefully one of those resolves your issue, but let me know if it doesn't and I can investigate further.

weegeeday, 2021-03-23:
My headset is plugged in, I am using oculus link, i can try using riftcat instead of link and also try riftcat on a different headset, riftcat is just like virtual desktop SteamVR but it was only made for VR Streaming, my readout is this: "407&303&247&239&771&511&511&0&0&0&0&0&0" i have one potentiometer plugged in at A4, when i turn it other values change, and i have changed my settings file.

weegeeday, 2021-03-23:
Good news! it does work, the only issue being that the hands are in the ground, also, is it possible to only have one lucidGlove hand enabled?

lucas-vrtech, 2021-03-24:
You can have only one hand enabled if you go in the deviceprovider.cpp and comment out the instantiation and runframe of one of the hands

lucas-vrtech, 2021-03-24:
If your hands are in the ground that may mean the controller discovery isn't going correctly. Are your controllers on and tracking correctly outside of SteamVR?

weegeeday, 2021-03-24:
yes, they work on oculus, and they work in the steamVR dashboard, so outside of games they can track, but then in games it doesnt want to track (in games that are supported by this driver)

weegeeday, 2021-03-24:
i could use ovr advanced settings i think to see if i can do something about it.

danwillm, 2021-03-24:
Interesting - so you built the `develop` branch and that fixed things?

There's a bug fix in there which prevents a crash when there isn't a headset connected which may have fixed that issue.

Have you configured the bindings for the games? You'll need to make sure you've set up the skeletal input and pose for the games you're testing with? There's some instructions here:

https://github.com/LucidVR/opengloves-driver/wiki/Configuring-SteamVR-Input

weegeeday, 2021-03-24:
alright great! Thanks for the help! I did do the binding for the fingers, but il do 100% of it to make it work. Il test out if binding the controls will fix my hands being in the ground but else it's working!

danwillm, 2021-03-24:
oh - yes, if you haven't set the bindings then that will be the issue :)

We've had some others who haven't set the bindings as well, so are obviously having problems. Probably a good idea to highlight that is setting up bindings is a required step, thanks for bringing it to our attention, will update the docs.

Let me know how it goes 👍

weegeeday, 2021-03-27:
it works great! i had a lot of trouble building it, but i have one finger done, and it works! thanks!

closed 2021-03-27 by weegeeday