LucidVR/opengloves-driver #40: Rel 0.1.1 changes controller icons in VR

timbomur, 2021-03-27:
Using 0.1.1 on a Win 10 and an Oculus Quest (not #2) changes controller icon in SteamVR home to what looks like a PS2 controller.  buttons no longer work.

lucas-vrtech, 2021-03-28:
Hi! Can you post a screenshot in here of what the icons look like? You may need to link the image from another site, or feel free to DM me on Discord LucidVR#0001

danwillm, 2021-03-28:
Hi,

This is expected in SteamVR home, the controllers (for now) should look like controllers.

Buttons on the touch controllers are also not expected to function - the idea is that you would implement your own buttons, joysticks etc. on the glove, which would then be the buttons that can be used in game.

To set bindings of these buttons, check out: https://github.com/LucidVR/opengloves-driver/wiki/Configuring-SteamVR-Input

danwillm, 2021-04-04:
Hey, it's been a week since this issue was opened - is it all working now?

danwillm, 2021-04-14:
I'm going to close this issue with the release of v0.2, but do feel free to reopen it or comment if you're still having problems.
https://github.com/LucidVR/opengloves-driver/releases/tag/v0.2

Thanks.

closed 2021-04-14 by danwillm