LucidVR/opengloves-driver #46: Force feedback integration

danwillm, 2021-03-29:
Integrate a force feedback interface between the driver and games.

Using OpenVR's `IVRIOBuffer` interface is probably the most suitable way to do this, but modding may also be necersarry.

https://github.com/ValveSoftware/openvr/wiki/IVRIOBuffer

kitlith, 2021-04-10:
it might be worth noting that that wiki page was updated 2 days ago to mark that interface as deprecated

relevant issue: https://github.com/ValveSoftware/openvr/issues/1530

danwillm, 2021-04-11:
I saw that as well, which is unfortunate. To my knowledge, there's no other way to achieve communication between applications and drivers directly through OpenVR other than that interface, which is a shame.

It would be interesting to know why it was deprecated.

kitlith, 2021-04-11:
well, perhaps you (or someone else, idk) should go talk to them on that issue (or open a new issue, idk) because they seemed open to going "hey, what do you need it for, we can point you in the right direction and/or work on a new api for it"

if i were more involved/didn't have too many other things on my plate i'd offer to help with that, but i'm not.

closed 2021-07-24 by danwillm