LucidVR/opengloves-driver #58: ESP as Controller

fharni, 2021-04-15:
Why not use an ESP8266 like Wemos D1 mini. It has a faster CPU and also Bluetooth build in.

gaming-se, 2021-04-18:
@fharni wrote
> Why not use an ESP8266 like Wemos D1 mini. It has a faster CPU and also Bluetooth build in.

Hi,
ESP8266 based controllers have only **one** Analog input pin:

1.  Most hardware Joysticks have a need of 2 Analog-pins for axis input (x and y)
2.  Fingers use currently for each finger one analog input  that would mean additional 5 Analog-pins if a resistor based sensor is used.
3.  If u think you would be able to use analog to digital converters then u get a bulkier device more wiring and its less beginner DIY friendly.

If an esp then i would suggest to buy one with enough analog pins like ESP32.

Wemos D1 mini **do not have Bluetooth** as far as i know, Could u provide the specs where u have seen one with ?
 
For myself i think i might give Arduino Nano 33 at later stages a try, at some point, not sure what is gonna be needed for later haptic feedback device.

Kind Regards

fharni, 2021-04-19:
Ok, you are right. I mixed it up with the ESP32. Then my question will change to? Why not use the ESP32 as a controller? It has 18 ADCs and a lot more pins. Also it has bluetooth and wifi. The processor can operate between 80 and 240 MHz and has enough memory on it. On AliExpress I will get one for 6 €. Maybe one controller can handle both hands.
https://microcontrollerslab.com/esp32-pinout-use-gpio-pins/

The Arduino nano 33 IOT costs 20 €. It has 8 ADCs, a clock speed of 48 MHz and 256 KB flash memory.
https://store.arduino.cc/arduino-nano-33-iot

gaming-se, 2021-04-19:
Hi,
As far as i know LucidVR developers are working on a ESP32 version.

I decided to go for myself for Adruno Nano 33 because it has a ATECC608A secure element integrated, it's a decision for myself. I kinda expect the special "ESP32-WROOM-32**SE**" to be hard to get I heard rumors that it not that easy but I never tried it so I guess (^_^) it's up to you to try it. (watch out not to get the ESP32-WROOM-32**U** or the ESP32-WROOM-32**D** version by mistake)

Kind Regards

danwillm, 2021-04-25:
Going to close this now that we are finalising use of the ESP32 for bluetooth.

closed 2021-04-25 by danwillm