LucidVR/opengloves-driver #63: missing .sln file

Kanken6174, 2021-04-23:
This makes your driver impossible to build from source. Remove the .sln file extension from the gitignore and that should fix it...

danwillm, 2021-04-23:
No, it doesn't. You can generate the project files from cmake included in the repository.

Please take a look at the [BUILDING.md](https://github.com/LucidVR/opengloves-driver/blob/develop/BUILDING.md) for how to generate the project files and how to build.

Please let me know if you have any further issues, I would be happy to help out.

Kanken6174, 2021-04-23:
Thank you! I completely forgot to use cmake even though there were cmake files in the repo...

Keep doing god's work with this project, and thanks for the quick response 😄

closed 2021-04-23 by Kanken6174

danwillm, 2021-04-23:
🚀