LucidVR/opengloves-driver #65: Build Problems

fharni, 2021-04-25:
I try to build this driver and got the following error messages:
`[build] E:\Quellcode\opengloves-driver\src\DeviceDriver\KnuckleDriver.cpp:11:1: error: a storage class can only be specified for objects and functions
[build]  static const enum ComponentIndex : int {
[build]  ^~~~~~
[build] E:\Quellcode\opengloves-driver\src\DeviceDriver\LucidGloveDriver.cpp:15:1: error: a storage class can only be specified for objects and functions
[build]  static const enum ComponentIndex : int {
[build]  ^~~~~~`
I build in Visual Studio Code with MinGw under Windows 10.

danwillm, 2021-04-25:
We use Microsoft Visual C++ compiler for building the project. Looking at the code, it looks like there was a wrong declaration with an enum - could you try using the `fix-gcc` branch I've just pushed, and let me know if that works for you?

fharni, 2021-04-25:
Great work. Now it is compiling successful.

danwillm, 2021-04-25:
Great.

Just to double check, was this using the new branch or using MS Visual C++ compiler?

fharni, 2021-04-26:
The new branch works with the MinGw compiler.

danwillm, 2021-04-26:
Cool, thanks for confirming. I'll merge these changes in.

closed 2021-04-26 by danwillm