LucidVR/opengloves-driver #85: Build and Link Problems

fharni, 2021-05-21:
I have problems building the current main v0.3 branch.
```
e:\Quellcode\opengloves-driver\build>cmake.exe -G "MinGW Makefiles" ..
-- The C compiler identification is GNU 8.1.0
-- The CXX compiler identification is GNU 8.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/mingw-w64/x86_64-8.1.0-posix-seh-rt_v6-rev0/mingw64/bin/gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/mingw-w64/x86_64-8.1.0-posix-seh-rt_v6-rev0/mingw64/bin/g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: E:/Quellcode/opengloves-driver/build

e:\Quellcode\opengloves-driver\build>mingw32-make
[  8%] Building CXX object CMakeFiles/driver_openglove.dir/src/Bones.cpp.obj
[ 16%] Building CXX object CMakeFiles/driver_openglove.dir/src/Communication/BTSerialCommunicationManager.cpp.obj
[ 25%] Building CXX object CMakeFiles/driver_openglove.dir/src/Communication/SerialCommunicationManager.cpp.obj
[ 33%] Building CXX object CMakeFiles/driver_openglove.dir/src/ControllerPose.cpp.obj
In file included from E:\Quellcode\opengloves-driver\src\ControllerPose.cpp:1:
E:/Quellcode/opengloves-driver/include/ControllerPose.h:23:7: error: extra qualification 'ControllerPose::' on member 'IsOtherRole' [-fpermissive]
  bool ControllerPose::IsOtherRole(int32_t test);
       ^~~~~~~~~~~~~~
mingw32-make[2]: *** [CMakeFiles\driver_openglove.dir\build.make:121: CMakeFiles/driver_openglove.dir/src/ControllerPose.cpp.obj] Error 1
mingw32-make[1]: *** [CMakeFiles\Makefile2:82: CMakeFiles/driver_openglove.dir/all] Error 2
mingw32-make: *** [Makefile:90: all] Error 2
```

I changed 
	`bool ControllerPose::IsOtherRole(int32_t test);`
to
	`bool IsOtherRole(int32_t test);`

After this, the linking process failes with this message:

```
Consolidate compiler generated dependencies of target driver_openglove
[  8%] Building CXX object CMakeFiles/driver_openglove.dir/src/ControllerPose.cpp.obj
[ 16%] Building CXX object CMakeFiles/driver_openglove.dir/src/DeviceDriver/KnuckleDriver.cpp.obj
[ 25%] Building CXX object CMakeFiles/driver_openglove.dir/src/DeviceDriver/LucidGloveDriver.cpp.obj
[ 33%] Building CXX object CMakeFiles/driver_openglove.dir/src/DeviceFactory.cpp.obj
In file included from E:/Quellcode/opengloves-driver/include/Communication/BTSerialCommunicationManager.h:14,
                 from E:/Quellcode/opengloves-driver/include/DeviceProvider.h:13,
                 from E:\Quellcode\opengloves-driver\src\DeviceFactory.cpp:4:
C:/PROGRA~1/MINGW-~1/X86_64~1.0-P/mingw64/x86_64-w64-mingw32/include/Winsock2.h:15:2: warning: #warning Please include winsock2.h before windows.h [-Wcpp]
 #warning Please include winsock2.h before windows.h
  ^~~~~~~
[ 41%] Building CXX object CMakeFiles/driver_openglove.dir/src/DeviceProvider.cpp.obj
In file included from E:/Quellcode/opengloves-driver/include/Communication/BTSerialCommunicationManager.h:14,
                 from E:/Quellcode/opengloves-driver/include/DeviceProvider.h:13,
                 from E:\Quellcode\opengloves-driver\src\DeviceProvider.cpp:1:
C:/PROGRA~1/MINGW-~1/X86_64~1.0-P/mingw64/x86_64-w64-mingw32/include/Winsock2.h:15:2: warning: #warning Please include winsock2.h before windows.h [-Wcpp]
 #warning Please include winsock2.h before windows.h
  ^~~~~~~
[ 50%] Building CXX object CMakeFiles/driver_openglove.dir/src/DriverLog.cpp.obj
[ 58%] Building CXX object CMakeFiles/driver_openglove.dir/src/Encode/LegacyEncodingManager.cpp.obj
[ 66%] Building CXX object CMakeFiles/driver_openglove.dir/src/Quaternion.cpp.obj
[ 75%] Linking CXX shared library libdriver_openglove.dll
CMakeFiles\driver_openglove.dir/objects.a(BTSerialCommunicationManager.cpp.obj):BTSerialCommunicationManager.cpp:(.text+0x3eb): undefined reference to `__imp_recv'
CMakeFiles\driver_openglove.dir/objects.a(BTSerialCommunicationManager.cpp.obj):BTSerialCommunicationManager.cpp:(.text+0x4b6): undefined reference to `__imp_shutdown'
CMakeFiles\driver_openglove.dir/objects.a(BTSerialCommunicationManager.cpp.obj):BTSerialCommunicationManager.cpp:(.text+0x4c9): undefined reference to `__imp_WSAGetLastError'
CMakeFiles\driver_openglove.dir/objects.a(BTSerialCommunicationManager.cpp.obj):BTSerialCommunicationManager.cpp:(.text+0x5a5): undefined reference to `BluetoothFindFirstDevice'
CMakeFiles\driver_openglove.dir/objects.a(BTSerialCommunicationManager.cpp.obj):BTSerialCommunicationManager.cpp:(.text+0x6df): undefined reference to `BluetoothFindNextDevice'
CMakeFiles\driver_openglove.dir/objects.a(BTSerialCommunicationManager.cpp.obj):BTSerialCommunicationManager.cpp:(.text+0x78e): undefined reference to `__imp_WSAStartup'
CMakeFiles\driver_openglove.dir/objects.a(BTSerialCommunicationManager.cpp.obj):BTSerialCommunicationManager.cpp:(.text+0x7eb): undefined reference to `__imp_socket'
CMakeFiles\driver_openglove.dir/objects.a(BTSerialCommunicationManager.cpp.obj):BTSerialCommunicationManager.cpp:(.text+0x872): undefined reference to `__imp_connect'
CMakeFiles\driver_openglove.dir/objects.a(BTSerialCommunicationManager.cpp.obj):BTSerialCommunicationManager.cpp:(.text+0x884): undefined reference to `__imp_WSAGetLastError'
CMakeFiles\driver_openglove.dir/objects.a(BTSerialCommunicationManager.cpp.obj):BTSerialCommunicationManager.cpp:(.text+0x8c0): undefined reference to `__imp_ioctlsocket'
CMakeFiles\driver_openglove.dir/objects.a(BTSerialCommunicationManager.cpp.obj):BTSerialCommunicationManager.cpp:(.text+0x92e): undefined reference to `__imp_send'
CMakeFiles\driver_openglove.dir/objects.a(BTSerialCommunicationManager.cpp.obj):BTSerialCommunicationManager.cpp:(.text+0x940): undefined reference to `__imp_WSAGetLastError'
CMakeFiles\driver_openglove.dir/objects.a(BTSerialCommunicationManager.cpp.obj):BTSerialCommunicationManager.cpp:(.text+0x962): undefined reference to `__imp_closesocket'
CMakeFiles\driver_openglove.dir/objects.a(BTSerialCommunicationManager.cpp.obj):BTSerialCommunicationManager.cpp:(.text+0x96b): undefined reference to `__imp_WSACleanup'
collect2.exe: error: ld returned 1 exit status
mingw32-make[2]: *** [CMakeFiles\driver_openglove.dir\build.make:271: libdriver_openglove.dll] Error 1
mingw32-make[1]: *** [CMakeFiles\Makefile2:82: CMakeFiles/driver_openglove.dir/all] Error 2
mingw32-make: *** [Makefile:90: all] Error 2
```

danwillm, 2021-05-21:
Hi,

Thanks for raising this - turns out msvc didn't catch that incorrect syntax.

I've gone ahead and updated the cmake and published `bugfix-85` branch to fix this and try the resolve the linking issues - can you let me know if that works for you?

The bluetooth implementation previously used `#pragma comments` to link the libraries, so I wouldn't have expected it to work with mingw, hopefully it should now.

fharni, 2021-05-21:
Thank you. Now I can compile and link the project, You can merge this branch and close this issue.

danwillm, 2021-05-21:
Glad to hear it we're sorry it was causing you issues in the first place.

closed 2021-05-21 by danwillm