LucidVR/opengloves-driver #93: Bluetooth not connecting / crashes

Nitwel, 2021-06-12:
When I try to use bluetooth, the driver crashes at this line:
https://github.com/LucidVR/opengloves-driver/blob/c039fdb0e105b99d3e37ee82da89171067a41022/src/Communication/BTSerialCommunicationManager.cpp#L198
It seems that it can't find the function `connect` or something.

Note that it is crashing and thus not even outputting `Could not connect socket to ESP32.` Which would be given out if the return value was not 0.

danwillm, 2021-07-20:
I'm going to close this for now, but feel free to re-open/comment if you still enounter issues with it.

Thanks.

closed 2021-07-20 by danwillm

danwillm, 2021-08-06:
We think that we've addressed this issue in a later commit, but feel free to re open if it doesn't work for you.