LucidVR/opengloves-driver #97: Allowing Controller Buttons to be Used While Using Gloves

Cozeque, 2021-06-30:
Is it possible to allow only controller buttons to remain active while using the gloves. For example, having the option to press the menu button on the controllers would be handy since there isn't one for the gloves.

danwillm, 2021-07-02:
We haven't really seen a use case where using the buttons on the proprietary controller is preffered to having buttons on the glove.

This isn't on the roadmap to be supported anytime soon, but let us know if you find a use case that makes this feature appropriate and we will consider adding it in.

closed 2021-07-02 by danwillm

danwillm, 2021-08-06:
If you're still interested in menu button functionality, this was added in a recent commit (https://github.com/LucidVR/opengloves-driver/commit/bd9e2852c848f48bcdd0b1704abe0da9327c7eae)