LucidVR/opengloves-driver #195: Failed to load driver_openglove.dll

breezZe, 2022-03-20:
**Problem**
SteamVR can't find gloves controller, two esp32 chips on two different serial ports. SteamVR log shows as `Unable to load driver openglove. Primary driver shared library not found on filesystem (for this architecture): C:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\OpenGlove\bin\win32\driver_openglove.dll.`

**Environment**
* Opengloves driver downloaded from Steam
* SteamVR, Steam, openXR runtime

I think there is no need to load a Win32 driver, while I've no idea how to deal with this.

[steamvr_settings.zip](https://github.com/LucidVR/opengloves-driver/files/8311467/steamvr_settings.zip)

danwillm, 2022-03-20:
We do not ship a 32-bit version of the driver, which is why you are seeing that error - there is nothing in `\bin\win32`, only `\bin\win64`.

Are you able to try it on a 64-bit pc?

breezZe, 2022-03-20:
> We do not ship a 32-bit version of the driver, which is why you are seeing that error - there is nothing in `\bin\win32`, only `\bin\win64`.
> 
> Are you able to try it on a 64-bit pc?

Thanks for quick responding. I'm quite sure my PC is 64-bit, and that's what confusing me. Maybe a SteamVR relavant issue? Furthermore, if `driver_openglove.dll` is loaded properly, will the openglove device  appear at the steamvr window?

danwillm, 2022-03-20:
> Maybe a SteamVR relavant issue? 

yes, this looks to be the case. If you're 100% sure you're on a 64-bit pc then you could try copying the .dll from \bin\win64 to \bin\win32 and see if that works.

> will the openglove device appear at the steamvr window?

yes, you'll see two skeleton-type hand icons appear in the status window.

breezZe, 2022-03-20:
> If you're 100% sure

yes I'm 100% sure.
> you could try copying the .dll from \bin\win64 to \bin\win32 and see if that works.

I've already done this, it shows ` Unable to load driver openglove from C:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\OpenGlove\bin\win32\driver_openglove.dll (193)`
and `ASSERT: "DriverManager openglove load error 193u" at c:\buildslave\steamvr_rel_win32\build\src\common\vrcommon\drivermanager.cpp:405.`

Is there any option to install or start steam as well as steamvr in 64-bit mode?  there must be something is wrongly configured.

danwillm, 2022-03-20:
Wait, did you download the driver from Steam? Your path indicates that it's attempting to load from the SteamVR drivers folder which means you must have pasted the files there? You do not need to do this, opengloves will link automatically to SteamVR when downloading.

Can you please delete it and try again?

Else, try uninstalling both OpenGloves and SteamVR fully and reinstalling (make sure nothing remains `C:\Program Files (x86)\Steam\steamapps\common\SteamVR`) and try again.

If that doesn't work, make sure you don't have a antivirus setup.

breezZe, 2022-03-21:
It works. What I did is updated SteamVR to the newest version, and removed `common\SteamVR\drivers\OpenGlove`. I suppose there was something wrong with my improper configuration. 
Thanks a lot.

danwillm, 2022-03-21:
Great, glad to hear that it's working.

closed 2022-03-21 by danwillm