LucidVR/opengloves-driver #201: Syncronising versioning

ShalokShalom, 2022-04-05:
So far as I can tell, the Gloves and this repo are not synced, when it comes to the version numbers. 

I think it can help to avoid confusion, if you use 0.5 for prototype 5, 0.6 for prototype 6 and so on

LittleNyanCat, 2022-04-05:
Opengloves is updated much more often than glove prototypes come out. Also opengloves works with other firmwares other than lucidgloves and supports features that not even lucidgloves supports (eg. splay tracking, per joint tracking)

danwillm, 2022-04-05:
Lucas and I have discussed this previously, and think that it's best if the gloves/driver don't sync with versioning. We think that the driver and lucidgloves hardware aren't tied directly to eachother - you don't have to build the lucidgloves to be able to use opengloves, and there's a ton of other community projects that use opengloves as well to interface with SteamVR.

The driver has grown to accommodate more hardware from the community than just the lucidgloves, and much of that hardware contains features that currently aren't present on the lucidgloves, such as splay, or haptic vibration support. So, having the versions synced wouldn't really be meaningful in this way.

The goal we have for the driver is to be able to make it so anyone who wants to make their own DIY hardware for VR can do so with relative ease, and we'll grow to accomodate features that these users want in the future, which might not be present on the lucidgloves hardware.

Hope that makes sense.

ShalokShalom, 2022-04-05:
I think your POV makes sense out of the perspective from you as a developer. 
Tell me, how many projects are **ACTUALLY** using and endorsing OpenGloves. 

Effectively, and out of the perspective of a user, this is the software for the Lucid Gloves.

> The goal we have for the driver is to be able to make it so anyone who wants to make their own DIY hardware for VR can do so with relative ease

Theory and practice are very different on this one. First, it would make most sense to also fork the gloves, but completely besides this: How does changing the versioning HURT that use case?

There is virtually no downside to this approach, but a major upside at an actual use case, instead of an envisioned one, that might or might not come into action. 

Currently, its just confusing for the majority of users, that come new to the project.

closed 2022-04-05 by ShalokShalom

danwillm, 2022-04-05:
>  Tell me, how many projects are ACTUALLY using and endorsing OpenGloves.

I can share statistics with you, and you can have a look for yourself on our discord server (https://discord.gg/lucidvr) - look at the glove-media channel. Even there, there's builds in the 100s of different versions of the glove, all which use OpenGloves.

Through steam we have >10,000 "users". It's obvious not all that have downloaded the driver have gloves built, but I expect a good majority of those users have built gloves.

The info we have available doesn't give a good estimate about how many people are currently using the driver - the realtime stats you can see on steamdb or we see on steamworks only account for people who have launched the UI, **not people who have the driver active**. The UI is only really required to setup the glove, and that you don't need it afterwards - we have ~50 daily users who launch the UI a day.

> Effectively, and out of the perspective of a user, this is the software for the Lucid Gloves.

Sure, but when it comes to it, I really don't think the majority of users are going to care what version the driver is on - it's recommended to download the driver on Steam, where we don't show versions. Only if you visit this repo, then go to the releases page would you see versions.

What happens if prototype 5 only improves on hardware and doesn't add any new features that require additions to the driver... does the driver jump bump up a version without changes? That doesn't follow most versioning practices...

We want to follow semantic versioning: we had some things we intentionally wanted to break to improve them (named pipes) from 0.4 -> 0.5, so we bumped the version up to show this. It doesn't make sense that these changes should be tied to prototype releases, especially with hundreds of users using the project for other purposes.

If I'm missing something let me know.

LittleNyanCat, 2022-04-05:
> Effectively, and out of the perspective of a user, this is the software for the Lucid Gloves.
That is very true, and I don't see the issue with that.

99% of people will be using the latest version of Opengloves from Steam, and grabbing whatever was the latest version of Lucidgloves when they built their gloves. Even when the firmware becomes inevitably outdated, both it and Opengloves have been maintained in a way that they are backward compatible (with the biggest exception being the jump from legacy encoding to alpha, and there's still an option to go around that). Even when new features are added to Opengloves that LucidGloves doesn't support (like splay or haptics), everything is honky-dory unless the user goes out of their way to enable them.

LittleNyanCat, 2022-04-05:
> Currently, its just confusing for the majority of users, that come new to the project.

Also just want to mention that we've had \*very few* signs of that. Worst case scenario someone manages to find an old version of LucidGloves with legacy encoding and tries to use it with alpha encoding on Opengloves (because alpha is the default) and it doesn't work. They go to our discord for help, and that gets sorted pretty quickly as taking a look at the output from the firmware is one of our main troubleshooting steps/tools. As someone who helps people get their gloves working on the daily, I can tell you the rate of this happening has been about 1/70, probably less.

We don't really feel the need to change to change how versioning is done

lucas-vrtech, 2022-04-05:
@ShalokShalom Hey there, just wanted to add to this.

> We want to follow semantic versioning: we had some things we intentionally wanted to break to improve them (named pipes) from 0.4 -> 0.5, so we bumped the version up to show this. It doesn't make [...]
Just wanted to emphasise this point a bit. No matter what prototype someone is building, they should be using the latest version of OpenGloves. Prototype 3 users should use the latest version, Prototype 4 users should be using the latest version. So in that respect, from said perspective of a user, it's actually better for the opengloves versions to be out of sync, because we don't want people to think that each version of the driver is tied to a prototype. In reality, every release makes *all* prototypes work better, regardless of what feature set is actually supported in the hardware. 

I'm not going to argue over whether or not OpenGloves is the software for LucidGloves, but what I will add is that without the people who are making their own custom versions of the gloves, the project wouldn't be nearly as developed as it currently is. The success isn't a one man show, it's a product of the amazing community that helped get it here.

I do really appreciate your concern on helping make it easier for people to follow the process of getting themselves set up. If you'd like to help this, something that would be much appreciated is to help clarify processes and steps in our wiki documentation. I couldn't do this extensively without help from everyone :)

ShalokShalom, 2022-04-06:
> @ShalokShalom Hey there, just wanted to add to this.
> No matter what prototype someone is building, they should be using the latest version of OpenGloves. Prototype 3 users should use the latest version, Prototype 4 users should be using the latest version. So in that respect, from said perspective of a user, it's actually better for the opengloves versions to be out of sync, because we don't want people to think that each version of the driver is tied to a prototype.

That makes sense, thanks!