LucidVR/opengloves-driver #202: Support for Linux

oxr463, 2022-04-05:
I'm running SteamVR on Ubuntu with my Valve Index.

CosmicFusion, 2022-04-24:
Yeah I would love to see this on Linux too... , And I'll imagine it to be a lot easier to implement than windows .

danwillm, 2022-04-29:
This is something I hope to work on this summer, including a port of opengloves to monado.

jurassicjordan, 2022-05-31:
> This is something I hope to work on this summer, including a port of opengloves to monado.

I cant want to see it. If you can make it compatible with ALVR (not certain if there needs to be any compatibility requirements) that would be majorly awesome

danwillm, 2022-09-09:
Opengloves is now supported in monado, here: https://gitlab.freedesktop.org/monado/monado/-/tree/main/src/xrt/drivers/opengloves

Will keep this issue open for now as I still plan on making the openvr driver linux compatible.