LucidVR/opengloves-driver #208: how to setup quest2

polo13999, 2022-08-24:
my arduino can  get data .I also setup config step by step.but it can't get my quest2 device .
I use virtual desktop

danwillm, 2022-08-24:
Please could you provide some more info on what exactly is the problem? Attaching logs would help: `C:\Program Files (x86)\Steam\logs\vrserver.txt`

polo13999, 2022-08-26:
Fri Aug 26 2022 15:02:51.897 - AUDIO: AudioDeviceChangeNotice (noting refresh): 2, device = {0.0.0.00000000}.{73f7f3cd-ed6b-4ea3-8f5b-4542a86551b0}
Fri Aug 26 2022 15:02:51.898 - AUDIO: OnDeviceStateChanged: DEVICE_STATE_UNPLUGGED (0x00000008), device = {0.0.0.00000000}.{73f7f3cd-ed6b-4ea3-8f5b-4542a86551b0}
Fri Aug 26 2022 15:02:51.898 - AUDIO: AudioDeviceChangeNotice (noting refresh): 4, device = {0.0.0.00000000}.{1fef3968-95fe-4880-9a38-c502a7bcf033}
Fri Aug 26 2022 15:02:51.898 - AUDIO: OnDefaultDeviceChanged: flow = eRender, role = eConsole, device = {0.0.0.00000000}.{1fef3968-95fe-4880-9a38-c502a7bcf033}
Fri Aug 26 2022 15:02:51.898 - AUDIO: AudioDeviceChangeNotice (noting refresh): 4, device = {0.0.0.00000000}.{1fef3968-95fe-4880-9a38-c502a7bcf033}
Fri Aug 26 2022 15:02:51.898 - AUDIO: OnDefaultDeviceChanged: flow = eRender, role = eMultimedia, device = {0.0.0.00000000}.{1fef3968-95fe-4880-9a38-c502a7bcf033}
Fri Aug 26 2022 15:02:51.898 - AUDIO: AudioDeviceChangeNotice (noting refresh): 4, device = {0.0.0.00000000}.{1fef3968-95fe-4880-9a38-c502a7bcf033}
Fri Aug 26 2022 15:02:51.898 - AUDIO: OnDefaultDeviceChanged: flow = eRender, role = eCommunications, device = {0.0.0.00000000}.{1fef3968-95fe-4880-9a38-c502a7bcf033}
Fri Aug 26 2022 15:02:52.008 - AUDIO: Refreshing audio devices
Fri Aug 26 2022 15:02:52.040 - AUDIO: Detected 18 playback and 29 record devices
Fri Aug 26 2022 15:45:04.927 - AUDIO: AudioDeviceChangeNotice (noting refresh): 2, device = {0.0.0.00000000}.{73f7f3cd-ed6b-4ea3-8f5b-4542a86551b0}
Fri Aug 26 2022 15:45:04.927 - AUDIO: OnDeviceStateChanged: DEVICE_STATE_NOTPRESENT (0x00000004), device = {0.0.0.00000000}.{73f7f3cd-ed6b-4ea3-8f5b-4542a86551b0}
Fri Aug 26 2022 15:45:05.010 - AUDIO: AudioDeviceChangeNotice (noting refresh): 2, device = {0.0.0.00000000}.{73f7f3cd-ed6b-4ea3-8f5b-4542a86551b0}
Fri Aug 26 2022 15:45:05.010 - AUDIO: OnDeviceStateChanged: DEVICE_STATE_ACTIVE (0x00000001), device = {0.0.0.00000000}.{73f7f3cd-ed6b-4ea3-8f5b-4542a86551b0}
Fri Aug 26 2022 15:45:05.011 - AUDIO: AudioDeviceChangeNotice (noting refresh): 4, device = {0.0.0.00000000}.{73f7f3cd-ed6b-4ea3-8f5b-4542a86551b0}
Fri Aug 26 2022 15:45:05.011 - AUDIO: OnDefaultDeviceChanged: flow = eRender, role = eConsole, device = {0.0.0.00000000}.{73f7f3cd-ed6b-4ea3-8f5b-4542a86551b0}
Fri Aug 26 2022 15:45:05.011 - AUDIO: AudioDeviceChangeNotice (noting refresh): 4, device = {0.0.0.00000000}.{73f7f3cd-ed6b-4ea3-8f5b-4542a86551b0}
Fri Aug 26 2022 15:45:05.011 - AUDIO: OnDefaultDeviceChanged: flow = eRender, role = eMultimedia, device = {0.0.0.00000000}.{73f7f3cd-ed6b-4ea3-8f5b-4542a86551b0}
Fri Aug 26 2022 15:45:05.011 - AUDIO: AudioDeviceChangeNotice (noting refresh): 4, device = {0.0.0.00000000}.{73f7f3cd-ed6b-4ea3-8f5b-4542a86551b0}
Fri Aug 26 2022 15:45:05.011 - AUDIO: OnDefaultDeviceChanged: flow = eRender, role = eCommunications, device = {0.0.0.00000000}.{73f7f3cd-ed6b-4ea3-8f5b-4542a86551b0}
Fri Aug 26 2022 15:45:05.094 - AUDIO: Refreshing audio devices
Fri Aug 26 2022 15:45:05.237 - AUDIO: Detected 18 playback and 29 record devices

I can't get any  response in opengloves with vr

danwillm, 2022-08-26:
This log doesn't show anything important and is only a small snippet of the entire thing. Please attach the full log file.

polo13999, 2022-08-29:
Mon Aug 29 2022 19:09:33.692 - //==============================================================================================
Mon Aug 29 2022 19:09:33.692 - ================================================================================================
Mon Aug 29 2022 19:09:33.692 - ================================================================================================
Mon Aug 29 2022 19:09:33.692 - vrserver.exe 1.23.7 startup with PID=9644, config=C:\Program Files (x86)\Steam\config, runtime=C:\Program Files (x86)\Steam\steamapps\common\SteamVR
Mon Aug 29 2022 19:09:33.694 - CSharedResourceNamespaceServer starting up
Mon Aug 29 2022 19:09:33.694 - Set thread FFFFFFFFFFFFFFFE priority to: 1
Mon Aug 29 2022 19:09:33.694 - WaitNamedPipe SteamVR_Namespace failed because no one is listening at that name.
Mon Aug 29 2022 19:09:33.694 - CSharedResourceNamespaceClient::Init(): failed connect
Mon Aug 29 2022 19:09:33.694 - CSharedResourceNamespaceClient::Init(): still waiting for vrserver to reply with namespace data
Mon Aug 29 2022 19:09:33.694 - Set thread FFFFFFFFFFFFFFFE priority to: -15
Mon Aug 29 2022 19:09:33.789 - Connecting client (SteamVR_Namespace) app container status: 1
Mon Aug 29 2022 19:09:33.789 - CSharedResourceNamespaceServer sent namespace data: 9644
Mon Aug 29 2022 19:09:33.804 - Client (SteamVR_Namespace) app container state: 1
Mon Aug 29 2022 19:09:33.820 - Connecting client (SteamVR_Namespace) app container status: 1
Mon Aug 29 2022 19:09:33.820 - CSharedResourceNamespaceServer sent namespace data: 9644
Mon Aug 29 2022 19:09:33.820 - CSharedResourceNamespaceClient::Init(): received namespace data 9644
Mon Aug 29 2022 19:09:33.820 - Self queried namespace: 9644
Mon Aug 29 2022 19:09:33.820 - VR server 1.23.7 (v1659988962) starting up with config=C:\Program Files (x86)\Steam\config
Mon Aug 29 2022 19:09:33.820 - VRServer will wait for VRMonitor before exiting
Mon Aug 29 2022 19:09:33.823 - [Settings] Load Default Json Settings from C:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\htc\resources\settings\default.vrsettings
Mon Aug 29 2022 19:09:33.823 - [Settings] Load Default Json Settings from C:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\lighthouse\resources\settings\default.vrsettings
Mon Aug 29 2022 19:09:33.823 - [Settings] Load Default Json Settings from C:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\null\resources\settings\default.vrsettings
Mon Aug 29 2022 19:09:33.823 - [Settings] Load Default Json Settings from C:\Program Files\Virtual Desktop Streamer\OpenVRDriver\resources\settings\default.vrsettings
Mon Aug 29 2022 19:09:33.823 - [Settings] Load Default Json Settings from C:\Program Files (x86)\Steam\steamapps\common\OpenGlove\resources\settings\default.vrsettings
Mon Aug 29 2022 19:09:33.824 - [Settings] Load Default Json Settings from C:\Program Files (x86)\Steam\steamapps\common\SteamVR\resources\settings\default.vrsettings
Mon Aug 29 2022 19:09:33.824 - [Settings] Load Json Settings from C:\Program Files (x86)\Steam\config\steamvr.vrsettings
Mon Aug 29 2022 19:09:33.827 - [Settings] Load Default Json Settings from C:\Program Files\Virtual Desktop Streamer\OpenVRDriver\resources\settings\default.vrsettings
Mon Aug 29 2022 19:09:33.827 - [Settings] Load Default Json Settings from C:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\htc\resources\settings\default.vrsettings
Mon Aug 29 2022 19:09:33.827 - [Settings] Load Default Json Settings from C:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\lighthouse\resources\settings\default.vrsettings
Mon Aug 29 2022 19:09:33.828 - [Settings] Load Default Json Settings from C:\Program Files (x86)\Steam\steamapps\common\OpenGlove\resources\settings\default.vrsettings
Mon Aug 29 2022 19:09:33.828 - [Settings] Load Default Json Settings from C:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\null\resources\settings\default.vrsettings
Mon Aug 29 2022 19:09:33.828 - [Settings] Load Default Json Settings from C:\Program Files (x86)\Steam\steamapps\common\SteamVR\resources\settings\default.vrsettings
Mon Aug 29 2022 19:09:33.828 - [Settings] Load Json Settings from C:\Program Files (x86)\Steam\config\steamvr.vrsettings
Mon Aug 29 2022 19:09:33.841 - Adding controller type vive
Mon Aug 29 2022 19:09:33.841 - Adding controller type vive_controller
Mon Aug 29 2022 19:09:33.841 - Adding controller type rift
Mon Aug 29 2022 19:09:33.842 - Adding controller type oculus_touch
Mon Aug 29 2022 19:09:33.842 - Adding controller type indexhmd
Mon Aug 29 2022 19:09:33.842 - Adding controller type knuckles
Mon Aug 29 2022 19:09:33.846 - Add Json firmware manifest from {htc}/firmware/manifest.vrfirmware
Mon Aug 29 2022 19:09:33.846 - Add Json firmware manifest from {indexcontroller}/firmware/manifest.vrfirmware
Mon Aug 29 2022 19:09:33.846 - Add Json firmware manifest from {indexhmd}/firmware/manifest.vrfirmware
Mon Aug 29 2022 19:09:33.846 - Add Json firmware manifest from {lighthouse}/firmware/manifest.vrfirmware
Mon Aug 29 2022 19:09:33.846 - Add Json firmware manifest from {tundra_labs}/firmware/manifest.vrfirmware
Mon Aug 29 2022 19:09:33.846 - Web language is tchinese
Mon Aug 29 2022 19:09:33.846 - [Web] Starting web server on port 27062
Mon Aug 29 2022 19:09:33.847 - Caching webfont C:\Program Files (x86)\Steam\steamapps\common\SteamVR\resources\webinterface\fonts\motiva-sans-black-italic.ttf.
[... cut ...]
Mon Aug 29 2022 19:09:39.179 - SetApplicationPid appkey=system.vrwebhelper.controllerbinding pid=1704, external transition
Mon Aug 29 2022 19:09:39.190 - SetApplicationPid appkey=system.systemui pid=13828, external transition
Mon Aug 29 2022 19:09:49.079 - Auto launch dashboard steam.overlay.1068820 has not started after 128 checks. Giving up waiting.
Mon Aug 29 2022 19:09:57.854 - oculus: Oculus touch 1PASH5D1P17365_Controller_Left controller state disconnected.
Mon Aug 29 2022 19:09:57.854 - oculus: Oculus touch 1PASH5D1P17365_Controller_Right controller state disconnected.

danwillm, 2022-08-29:
you have opengloves disabled in settings. Go to SteamVR settings > startup/shutdown > manage addons and enabled opengloves.