LucidVR/opengloves-driver #209: Cmake: Could not find a package configuration file provided by "nlohmann_json"

WillNilges, 2022-11-10:
Getting an error at the `cmake ..` step in the build instructions:

```
CMake Error at driver/overlay/CMakeLists.txt:1 (find_package):
  Could not find a package configuration file provided by "nlohmann_json"
  with any of the following names:

    nlohmann_jsonConfig.cmake
    nlohmann_json-config.cmake

  Add the installation prefix of "nlohmann_json" to CMAKE_PREFIX_PATH or set
  "nlohmann_json_DIR" to a directory containing one of the above files.  If
  "nlohmann_json" provides a separate development package or SDK, be sure it
  has been installed.

```

Guessing a new package or submodule was added?

WillNilges, 2022-11-10:
Looking at this to try to fix it: https://github.com/nlohmann/json/issues/1573

danwillm, 2022-11-17:
I think you've figured this out already, but you need to have vcpkg installed to get these libraries 😁

closed 2022-11-17 by danwillm