LucidVR/opengloves-driver #211: Support for macos

spotandjake, 2022-11-14:
I'm running SteamVR on Macos with my Oculus quest, and was wondering if there is any plan to support macos with this driver?

Stephen-3DVR, 2022-11-14:
A) Not an issue
B) How is that possible? Valve abandoned Steamvr on macOS years ago, and I don't believe Oculus ever supported it

danwillm, 2022-11-14:
I don't think we even have the ability to support macos (I'm not sure how you've got it working with Oculus in the first place) - we would have to move the driver back to older interfaces that are compatible with the build for macos steamvr, which isn't really in our interested to do.

We have support for linux coming soon, though, as that is a supported platform for steamvr.

closed 2022-11-14 by danwillm