LucidVR/opengloves-driver #213: Support for Splay force feedback

newbie-honnicutt, 2022-11-17:
I know that force feedback has been implemented, but it only prevents the fingers from curling more. Imagine you were holding a pen between two of your fingers; it would be useful to prevent those fingers from splaying even further.

danwillm, 2022-11-17:
If you have hardware that supports this, then please do share it and we can implement this functionality into opengloves.
We tend to implement features as and when community members come up with hardware that needs it.

newbie-honnicutt, 2022-12-16:
Ok, so because I don't have the equipment for testing my idea, I'm simply going to share an image that should explain it.

https://imgur.com/a/ftW1Wjj