LucidVR/opengloves-driver #219: Allow Driver to set hand position without controllers

krazysh01, 2023-01-19:
Allowing the driver to recieve position directly instead of using a controller offset would allow the use of onboard hand tracking on devices like the Quest 2 or Pico 4

danwillm, 2023-01-19:
Do you have a specific use case in mind where this would be helpful?

krazysh01, 2023-01-19:
Right now using Lucid Gloves or any other Haptic Gloves require the user attach controllers to the gloves/arms which adds alot of weight (unless using vive trackers which requires base stations) however most standalone headsets support hand tracking which is robust (to the point of even tracking hands in gloves) so being able to directly use their data (forwarding it from the headset and using named pipes to send the data to the driver) would allow lighter glove designs like the tactgloves from bhaptics.

krazysh01, 2023-01-19:
for a more specific use case, right now on Quest 2 using either Virtual Desktop or Oculus Airlink you can't use the hand tracking on PC, however there have been projects to as a background process read the hand tracking data and transmit it but there's no easy way to use it on PC (right now the primary use has been feeding it into OSC in VRChat) being able to feed the hand position data alongside finger position into OpenGloves would allow much more versatility.