LucidVR/opengloves-driver #235: Controller without gloves

karizr45, 2023-08-02:
Hey can I use this driver without gloves? 
If I can them can you please help me write sketch for arduino

NordGeit, 2023-12-16:
_Let's make some noise..._

Hey, I've been hankering and WANTING SO BAD FOR SEVERAL YEARS to see just... Any normal SteamVR tracked controller with a joystick and normal buttons for once. (Index unavailable here, no, not gonna import.)

Can? [There is a project](https://github.com/altunidev/SteamVR-Controller), but it's going nowhere right now. Any help would be appreciated and I am getting midnight madness over finding nothing I could harvest from (though that may be my own ineptitude) from the nine years of Vive existence.

danwillm, 2023-12-16:
Hi,
You can adapt opengloves to do this, yes. It supports communicating with a device over usb, bluetooth or named pipes and feeding the data it gets from that into steamvr. https://github.com/LucidVR/opengloves-driver/wiki/Driver-Input
However, it does not support doing tracking itself, and if that's something you're looking for then I'd recommend checking out the SteamVR Tracking HDK - https://partner.steamgames.com/vrlicensing