LucidVR/opengloves-driver #241: SteamVR Error 301 when Left Hand disabled with Named Pipes enabled

KingOfDranovis, 2023-12-25:
I was able to figure this out today, but it sure wasn't fun to deal with
This is something that should probably get fixed, as if you don't know it's happening then it can be harder to figure out. The only reason I figured out how to fix it was searching and finding the same problem in the Discord server.
Luckily I found something from danwillm that said to delete the vrsettings (C:\Program Files (x86)\Steam\config\steamvr.vrsettings) and so I did then validated the files, and now it seems to be working, and if I was to disable the left hand again, it stops working.

danwillm, 2023-12-26:
Hi,
I'm unfortunately not able to repro this issue. Do you have any additional steps needed to get this crash?
If you need any more help with, you're free to message me on discord: `danwillm`.

Thanks

KingOfDranovis, 2023-12-26:
Huh that's strange
Well I first got the error when attempting to use it along with Driver4VR but even after uninstalling that the error still happened, to the point that reinstalling SteamVR didn't stop the error from happening, which I realize was due to the cloud backup of the vrsettings file. What fixed it was deleting the vrsettings file, but after that disabling the left glove, which I had disabled before since I was only going to test with the right glove, still crashed SteamVR. It being a reinstall and a new vrsettings file made me think that it was due to the left glove, although I suppose there could be other parts at play that I'm not aware of due to the cloud backup making it not a fully fresh install of SteamVR.