LucidVR/opengloves-driver #244: Implement BLE

DieKatzchen, 2024-04-01:
The ESP32S3 doesn't support classic bluetooth, so I went ahead and implemented BLE to the firmware (PR [#123](https://github.com/LucidVR/lucidgloves/pull/123).) I'm using the Nordic UART Service standard, which is a Service with UUID 6E400001-B5A3-F393-E0A9-E50E24DCCA9E, a characteristic 6E400002-B5A3-F393-E0A9-E50E24DCCA9E that can be written, and a characteristic 6E400003-B5A3-F393-E0A9-E50E24DCCA9E that can be read with notify. I'm currently using an unholy kludge of virtual serial ports that requires that device signing be turned off, so if this could be implemented into the driver, I could stop doing that.

memmam, 2024-04-21:
In the meantime until this can be implemented I'd appreciate if you could provide details on said kludge, I want to build a mocap-focused Fae Mod variant to bridge from OpenGloves to VMC for use with motion capture software.

DieKatzchen, 2024-04-21:
It actually never worked right, the instant FFB was turned on it would crash the virtual serial port. I ended up making a simpler solution. [This code](https://github.com/DieKatzchen/Nordic-UART-Service-Bridge) can be flashed to a spare ESP32 and will act as a dongle.